adafruit/Adafruit_FONA #25: ignore unsolicited

ladyada, 2015-06-10:
unsolicited messages are annoying and confuse the parser - so maybe have a flag in readline() that if its an unsolicited message, retry the read?

theicfire, 2015-11-02:
You can tell if it's a URC by turning on AT+CFGRI=1 and then watching the RI pin. I'm not sure how to do it without having a big table of URC's otherwise.

theicfire, 2015-11-03:
Revision: It looks like `AT+CFGRI=1` only changes the RI pin in a small subset of URC's. Additionally, the occurrence of the RI pin and the URC coming over serial can be tricky to match up. For example, if SoftwareSerial is used, there's not much of a way to tell if the current message was sent during an RI change. Beyond this, there are references in the documentation that specifies URC's do not have to happen when the RI pin is down.

What this means is that the RI pin should only be used to.. do things that aren't related to URC's.

Currently, I see two solutions to solving this:
1) In readline, put in something like this:
```
        if (!multiline) {
          if (memcmp("+CMTI", replybuffer, 5) == 0) { // Check that the message is any of the URC's
            Serial.println("+CMTI skipped");
            timeout = 5000; // Increase the timeout somehow
            replyidx = 0;
            continue;
          } else {
            timeout = 0;
            break;
          }
        }
```
2) Go through section 19.3 of the command manual, and turn off all that's possible of the URC's in setup. Two examples:
```
        if (!multiline) {
          if (memcmp("+CMTI", replybuffer, 5) == 0) {
            Serial.println("+CMTI skipped");
            timeout = 5000;
            replyidx = 0;
            continue;
          } else {
            timeout = 0;         // the second 0x0A is the end of the line
            break;
          }
        }
```

dmayhem, 2016-02-26:
I would not say ignore them, I would say process them apropriately. Unsolicited messages are usually high-importance things like incoming calls, incoming messages, etc