adafruit/Adafruit_FONA #28: Error compiling

vrajput, 2015-06-23:
Just downloaded FONA_Library, 
Getting error in compilation

In member funtion 'boolean Adafruit_FONA::TCPsend(char*, uint_t);
[image]

jkeefe, 2015-06-24:
Me, too. Running Arduino IDE 1.0.5 on a Mac. Same error, same line.

ladyada, 2015-06-24:
Hiya, can you try 1.6.4?

toddtreece, 2015-06-24:
It looks like this is related to older versions of the IDE. The issue is fixed in v1.0.6+ of the IDE, so upgrading to the latest version (v1.6.5) should fix the compile errors you are seeing.

[image]

Relevant changes from the Arduino IDE changelog:

```
ARDUINO 1.0.6 - 2014.09.16

[core]
* avr: Improved USB-CDC write speed (Justin Rajewski)
* avr: Improved USB-CDC read code (Paul Brook)
* avr: Fixed race condition in USB-CDC transmit (Paul Brook)
* avr: Added replacement stub for cstdlib atexit() funciton (Christopher Andrews)
* Fixed wrong NULL pointer handling in Stream class (Amulya Kumar Sahoo)
* Backported String class from IDE 1.5.x (Matt Jenkins)
* Backported Print class from IDE 1.5.x
* Backported digitalPinToInterrupt macro from IDE 1.5.x
* Added initVariant() hook to allow 3rd party variant-specific initialization
* Added replacement stub for cstdlib atexit() function (Christopher Andrews)
```

closed 2015-06-24 by toddtreece

jkeefe, 2015-06-24:
Thanks! Will try that!