adafruit/Adafruit_FONA #39: getGPS buffer copy length

jychuah, 2015-10-04:
Adafruit_FONA.cpp line 712:

uint8_t len = max(maxbuff-1, strlen(p));

Should call to max be a call to min? Example call to buffer uses a length of 80. Data returned is length 89. This call will copy 89 characters to an 80 character buffer.

tdicola, 2015-10-06:
Ah great point, thanks for raising the issue.  Hrm I think it's tricky to switch to min since you might get a partial result instead of a full GPS response.  Overflowing the buffer is not good either, so it's a bit of a tradeoff.  I think the best option for now is to make sure the buffer passed in is large enough to hold the GPS response.  In the overload of getGPS (that parses out the response) it uses 120 characters (see https://github.com/adafruit/Adafruit_FONA_Library/blob/master/Adafruit_FONA.cpp#L722), so I just bumped up the buffer size in the example to be the same.  Thanks for raising the issue!

closed 2015-10-06 by tdicola

cinderblock, 2017-01-28:
Why was this issue closed? This seems like a major bug to me.

If I tell `getGPS()` it only has room to put `x` `char`s into ram location `m`, it better not put stuff outside the range of `[m, m+x]`.

This makes it really easy to break your own code.