adafruit/Adafruit_FONA #42: getSMSInterrupt is improperly named

theicfire, 2015-11-02:
`AT+CFGRI=1` makes the FONA look to see if an Unsolicited Result Code (URC) should trigger the RI pin low for 120 ms. It doesn't have much to do with SMS's.

ladyada, 2015-11-02:
yah but SMS's are URCs

theicfire, 2015-11-02:
Hmm, but regardless of the setting for `AT+CFGRI`, RI will go low on an SMS

ladyada, 2015-11-02:
huh really? its been a year but i recall needing to add that in order to get a pulse on the RI line. ok well, not sure when we'll rename it or to what but next time we do a sweep ill take a look!

theicfire, 2015-11-02:
[image]

theicfire, 2015-11-03:
A better table:

[image]

rickjansen, 2016-04-29:
(Watch out for the spelling of the under-voltage warnings, with double-NN.)

The SIM800 series have a nice alarm clock (AT+CALA), it would have been nice if an alarm expiry also triggered a RI. Now the alarm clock cannot wake a SIM800 that is in sleep mode (slow clock enabled), because the serial connection is down.