adafruit/Adafruit_FONA #45: Dynamically setting GPRS APN Settings

WillJonesMagic, 2016-01-12:
APN Settings appear to need to be burned into the code at runtime and don't support being set from variables or values stored in the EEPROM. I need to be able to send an SMS containing the APN settings to the device and have it store the values and have them persist through a reboot.
There's quite possibly something obvious I'm missing here but I've not been able to solve this one.

feinstein, 2016-01-22:
@WillJonesMagic I have a problem like yours in my project, so I made a modification in the library so you can pass the APN settings both by burning it into code or by calling a function, it's on my pull request #50. 

This way you can send an SMS, read it, parse it in code, and pass it to FONA.

WillJonesMagic, 2016-02-02:
@feinstein I tried your code, but the GSMLoc function appears to be broken. I get failure code 601 whenever I try to query the GSMLoc. GPS works fine though.

feinstein, 2016-02-02:
@WillJonesMagic I didn't change anything on GSMLoc, does it work using the `master` branch's code?

feinstein, 2016-02-03:
@WillJonesMagic I have fixed the code and tested it here, can you confirm if it's working for you?

WillJonesMagic, 2016-02-03:
@feinstein Perfect! Thank you!