adafruit/Adafruit_FONA #46: getNumSMS failed with FONA 800 uFL

king-jam, 2016-01-15:
Recently purchased a module  and found that:

        if (! sendParseReply(F("AT+CPMS?"), F("\"SM\","), &numsms) ) return -1;
      } else {
        if (! sendParseReply(F("AT+CPMS?"), F("+CPMS: \"SM_P\","), &numsms) ) return -1;
      }
https://github.com/adafruit/Adafruit_FONA_Library/blob/master/Adafruit_FONA.cpp#L483

needs to be changed to:

        if (! sendParseReply(F("AT+CPMS?"), F("\"SM\","), &numsms) ) return -1;
      } else {
        if (! sendParseReply(F("AT+CPMS?"), F("+CPMS: \"SM\","), &numsms) ) return -1;
      }

I don't know the scope of this issue or if I have a one off firmware rev on the SIM800 but might be worth looking into.

ladyada, 2016-01-15:
do you have a SIM800L or SIM800H? can you check?

king-jam, 2016-01-15:
Apologies, SIM800H.

ladyada, 2016-01-17:
mind grabbing the very latest lib ad trying again? we now check for SM_P as well as SM

king-jam, 2016-01-19:
Tested and verified it is working. Thanks for the fix!

closed 2016-01-19 by king-jam