adafruit/Adafruit_FONA #51: GPS Fix on FONA 3G

ValentmSTEM, 2016-01-22:
The GPS test and FONA test 3D location fix hasn't worked for 2 different 3G FONA boards I've purchased (with active antennae). Is this just a code issue with the 3G? The SMS and everything else is working fine. Where are the AT commands given as they aren't in the sketch?

ladyada, 2016-01-22:
hiya, GPS commands are here
https://github.com/adafruit/Adafruit_FONA_Library/blob/master/Adafruit_FONA.cpp#L717
and the AT command set is here
https://learn.adafruit.com/adafruit-fona-3g-cellular-gps-breakout/downloads
if you are having *tech support* issues please post up on the adafruit tech support forums. thx!

closed 2016-01-22 by ladyada

OscarVanL, 2020-03-23:
Did you manage to get GPS test to work for the Fona 3G?