adafruit/Adafruit_FONA #52: blockings when using the function "callerIdNotification"

UdFrederic, 2016-01-27:
Hello! I have recently bought a FONA shild V2 with SIM R14.18 and I am encountering blockings when using the function "callerIdNotification" with last code (version of Jan. 21st, 2016).
Could you please check or give any advice? Thank you very much!
Regards, UdFrederic

ladyada, 2016-01-28:
hiya please post all of the DEBUG output so its clear what exactly is happening

UdFrederic, 2016-02-22:
First, I have to point pout, that the FONA shield in generally is performing fine.  The FONAtest loads fine, I can get and do telephone calls and the GPS is running fine.

My goal is to have a continous GPS logging, as soon as a call is detected some alternative action shall be performed (in my case sending an e-mail with GPS coordinates.

Therefore I was doing many tests with the „IncomingCall“ example of the FONA library.
Therefore, as I understand the initial comments and the code correctly, I have to connect the RI with the digital pin 2, what I did. But then, when uploading the code and opening the serial monitor, the initialisation of the shield fails. Here an excerpt from the serial output:

`FONA incoming call example
Initializing....(May take 3 seconds)
Attempting to open comm with ATs
	---> AT
	<--- 
	---> AT
	<--- 
(…... and many Ats more)
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
Timeout: No response to AT... last ditch attempt.
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> ATE0
	<--- 
	---> ATE0
	<--- 
Couldn't find FONA`

Running the same code again but with removed connection between RI and digital pin 2 the FONA shild initializes smoothly:

`FONA incoming call example
Initializing....(May take 3 seconds)
Attempting to open comm with ATs
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- ÿ
	---> AT
	<--- AT
	---> AT
	<--- AT
	---> AT
	<--- AT
	---> AT
	<--- AT
	---> AT
	<--- AT
	---> ATE0
	<--- ATE0
	---> ATE0
	<--- OK
	---> AT+CVHU=0
	<--- OK
	---> ATI
	<--- SIM808 R14.18

OK

FONA is OK
> Incoming call...
	---> AT+CLIP=1
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
	<--- OK
Caller id notification enabled.`

When I then place a call I get:

`Phone Number: +CLIP: "+49xxxxxxxxxx",145,"",0,"",0
RING!
Phone Number: + 49xxxxxxxxxx`

For confidentiality of my tel number I replaces the real number by a serie of x, so the call was properly detected. This only works one time, so a second call will not be registered again. I assume this is done because the code was not ment to run without the RI non-connected.

So actually I deduce that I can not use the "IncomminCall" example because it seems to not be running find. Can you follow this argumentation or is there any error in my thinking?

UdFrederic, 2016-02-22:
First, I have to point out, that the FONA shield in generally is performing fine (Arduino UNO R3). The “FONAtest“ code loads fine, I can get and do telephone calls and the GPS is also running fine.

The goal for my project is to have a continuous GPS logging, as soon as a call is detected some alternative action shall be performed (in my case sending an e-mail with GPS coordinates).

Therefore I was doing many tests with the “IncomingCall“ example of the FONA library.
As understand the initial comments in the code as well as the code itself, I have to connect the RI with the digital pin 2, what I did. But then, when uploading the code and opening the serial monitor, the initialisation of the shield fails. Here an excerpt from the serial output:
```
FONA incoming call example
Initializing....(May take 3 seconds)
Attempting to open comm with ATs
	---> AT
	<--- 
	---> AT
	<--- 
_(... and many ATs more which I skip for readability)_
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
Timeout: No response to AT... last ditch attempt.
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> ATE0
	<--- 
	---> ATE0
	<--- 
Couldn't find FONA
```
Running the same code again but with removed connection between RI and digital pin 2 the FONA shild initializes smoothly:

```
FONA incoming call example
Initializing....(May take 3 seconds)
Attempting to open comm with ATs
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- ÿ
	---> AT
	<--- AT
	---> AT
	<--- AT
	---> AT
	<--- AT
	---> AT
	<--- AT
	---> AT
	<--- AT
	---> ATE0
	<--- ATE0
	---> ATE0
	<--- OK
	---> AT+CVHU=0
	<--- OK
	---> ATI
	<--- SIM808 R14.18

OK

FONA is OK

> Incoming call...
	---> AT+CLIP=1
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
> Incoming call...
	<--- OK
Caller id notification enabled.

`````

When I then place a call, I get:

```
Phone Number: +CLIP: "+49xxxxxxxxxx",145,"",0,"",0
RING!
Phone Number: + 49xxxxxxxxxx
```

For confidentiality of my tel number I replaced the real number by a series of x. So the call was properly detected. 

This only works one time, so a second call will not be registered again. I assume this is done because the code was not ment to run with RI non-connected.

So actually I can not use the "IncomingCall" test code as I thought it was ment to work. Can you agree on this or is there any wrong thinking on my side?

malachib, 2016-02-23:
Might be unrelated, but your extended "AT" initialization stuff could be an artifact of undervoltage/underpower situation.  Mine does that if I don't juice it properly

ladyada, 2016-02-24:
yah you may need to not use the API but instead hand-parse the response since you can see the "Incoming call..." messages are getting in the way of the send/receive code.

hosa12m, 2016-12-14:
sorry but I need help in another thing 
if I receive call  I need fona auto reply the call if the number I received was stored before on the sim 
or the fono  close line 
sorry if my English is not good