adafruit/Adafruit_FONA #53: Make DebugStream configurable

thedarkman, 2016-01-29:
Hi there, if i use the library with only a FONA connected, everything works good. But, as a arduino nano has only one hardware serial and i want to use that for external gps, i have to switch off debugging in FONAConfig.h. This works ofcause, but without debug informations it's difficult to program and sadly my C is not so good so i could not change the DebugStream to a SoftwareSerial my sketch creates for debugging.

Can anybody help here?

ladyada, 2016-02-01:
for the external GPS you really need to use another SoftwareSerial...or use a FONA808 which has GPS integrated. yeah...this is the side effect of only one hardware serial port. A 32u4 or ATSAMD arduino would help too, since you gain an extra serial console!

closed 2016-02-01 by ladyada

thedarkman, 2016-02-01:
Yeah, the Fona808 would be great, but i need special gps module.

Solved my problem with using a second SoftwareSerial and only check gps when needed. Otherwise i can't get notifications from Fona.