adafruit/Adafruit_FONA #58: HTTP support for FONA 3G

rogerguess, 2016-02-28:
Please!

simonmags, 2016-06-21:
Not sure if you have seen the m2msupport module, but you can do http requests through that and I think you can do some basic stuff through the fona test scripts. If you need more you should have a look at something like Temboo.

savire001, 2016-08-08:
Need to work with cloud like dweet.io and my 3G, but HTTP isn't supported yet because of different AT Command Set of SIM808 and SIM5320.
help please!

killaimdie, 2016-11-11:
[http://simcom.ee/documents/?dir=SIM5320](url) is the documentation folder for the SIMCOM 5320A, you can see that simcom has included a SIM52xxx HTTP application note, I assume because the AT commands therein are supported. If you get it working please post some code on the official Fona forums because a lot of folks are having this issue.

simonmags, 2017-02-20:
We are working on this and are getting closer to a resolution, discussions are ongoing on this forum if you can join in and help:

[(https://forums.adafruit.com/viewtopic.php?f=54&t=109601&p=560531#p560531)]

TaylorHawkes, 2017-07-13:
I got basic HTTP /GET Requests working on my fork of this repo: 
https://github.com/TaylorHawkes/Adafruit_FONA 

Just view my first commit on the repo to see what I did... I suck at c++ and just whipped solution up, so this code still need a lot of work,  but it may help you guys get going... @simonmags  - let me know if you want me to open PR.

Paulitam, 2017-07-26:
I am trying to send a POST request by using a SIMCOM5320E and an arduino uno, but I get a Bad Request Error. Please, could yo help me with ideas about where are my mistakes. Following the instructions of the manufacturer, and taking into account that I am connected to the net GSM or WCDMA, my program goes:

AT+CHTTPSSTART

OK
AT+CHTTPSOPSE="www.myhttpserver.com", 80, 1

OK
AT+CHTTPSSEND=207

>
POST /xxxxxxxxx/xxx/xxx.php/xxx/xxxxxxxxxxx HTTP/1.1\r\nHost: xxxxxxxxxxxxxxxxxx.xxxxxxxxx.xxxxxxx.xxxxxxxxx.com\r\nContent-Length: 73\r\n\r\nmatricula=5434543&macdispositivo=0245874596&estado=PRODUCIENDO&estadoId=1\x1A
OK
AT+CHTTPSSEND

OK

+CHTTPSSEND: 0

+CHTTPS: RECV EVENT

+CHTTPSNOTIFY: PEER CLOSED
AT+CHTTPSRECV=450

OK

+CHTTPSRECV: DATA,419
HTTP/1.1 400 Bad Request

simonmags, 2017-08-21:
If you want support try heading over to a forum like the Adafruit forum or the Arduino forum, you will get more help there, this isn't the place to debug code

congon4tor, 2018-04-19:
Any update on how to get http working with the fona 3g?? I've been reading the adafruit forum but have not seen a clear solution to this issue.

amirfarazmand, 2019-02-17:
Even their own commands with Arduino is not working. Did anyone has HTTP working on this board?

congon4tor, 2019-02-17:
No I ended up using the other FONA board

amirfarazmand, 2019-02-17:
What board is it mate?

congon4tor, 2019-02-17:
FONA 808

amirfarazmand, 2019-02-17:
Thats 2G though. There is not support in Australia.

congon4tor, 2019-02-17:
There was a post in the adafruit forum about http in the Fona 3G but I doubt it will give any solutions it didn't 6 months ago.

amirfarazmand, 2019-02-17:
Their support was hopeless. Kept repeating the question without reading any replies.

ladyada, 2019-02-17:
the 3G doesn't have a built in HTTP command set, that's why it isn't supported

amirfarazmand, 2019-02-17:
Fona 3G has option to read a website that doesnt work. Not sure what you mean by it doesnt?

AskMP, 2019-10-18:
Has there been any updates here? I'm able to use AT+CHTTPSOPSE and subsequent commands using comm but it's just errors coming from the library.