adafruit/Adafruit_FONA #60: TCPread length of data read can be bigger than 256 ?

bilics, 2016-03-01:
Hi,

The TCPread call is defined as:

`uint16_t Adafruit_FONA::TCPread(uint8_t *buff, uint8_t len)`

what happens when you have more than 256 bytes to read ? Or that never happens ?

Another thing is that 

`uint16_t Adafruit_FONA::TCPavailable(void)`  

is able to return a length greater than 256 to read from... should we read 256 bytes at a time ?

Thanks!

feinstein, 2016-06-07:
AFAIK the SIM module can give up to 1460 bytes of data in a sigle `read command`. 

**BUT** the FONA library limits it to 256 bytes at a time. I believe it was done this way so you can have a static memory place for the received bytes (so no `malloc()` and all the problems that comes with it) and still not be too much to suck all the Arduino's RAM (which is around 2k).

feinstein, 2016-06-21:
I just sent some fresh code that can have a bigger buffer than the fixed 255 bytes...now the user have to provide a buffer and it's size at the constructor.