adafruit/Adafruit_FONA #89: setBaudRate sends invalid value for 115200

mschroebel, 2017-10-02:
Working with an Arduino Uno and a FONA 3G. I was following instructions to set baud to 4800. But thought, gosh, I remember when 9600 baud was fast, 4800? So I looked a little harder and noticed this error message in the FONA3G_setbaud sketch:
---> AT+IPREX=49664
Delved further and method ::setBaudRate has baud defined as unsigned int 16 which isn't large enough to store 115200 (only 65535). I change to int32_t and seems to be fine now. Probably ought to be a string and maybe improved with further validating values, the 5320 will provide a list of valid values with AT+IPREX?

simonmags, 2017-10-04:
will you add that fix into the main repo?

joeinaustin, 2019-09-08:
Also ran into this.  Simple int32_t fix should be added to main.