adafruit/Adafruit_FONA #104: fona.GPRSState reports enabled when GPRS has not yet been enabled

chenein, 2019-05-20:
[FONAtest_gprs.txt](https://github.com/adafruit/Adafruit_FONA/files/3200022/FONAtest_gprs.txt)

The fona.GPRSState command returns true as soon as the network status changes from a disconnected to a connected state, even if the fona.enableGPRS(true) function has not yet been called.

The issue may be in the modem itself, as the AT+CGATT? command returns 1 in this case when it should not. 

This issue prevents me from making a sketch that automatically reconnects GPRS if it unexpectedly goes down. This is because the fona.GPRSState function doesn't accurately report when GPRS is up and down.

I am attaching a version of the FONAtest sketch that has an extra command (D) to run fona.GPRSState. It can be used to validate this bug. You will see that 'D' returns '0' while the modem is trying to connect to the network. However, upon connecting, 'D' begins returning '1' even though the 'G' (Enable GPRS) command has not yet been run. 

- Arduino board:  UNO
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.3

jegjessing, 2019-11-09:
Any news on this?

chenein, 2019-11-27:
I haven't heard anything. Still interested in whether this could be fixed though.