adafruit/Adafruit_FONA #111: readRTC not finished

bt20304, 2020-01-30:
https://github.com/adafruit/Adafruit_FONA/blob/4a556e180fdeefb9518fac160a8abf915ae8c92c/Adafruit_FONA.cpp#L153

The function was made, and an attempt to parse was complete, but never finished. It would be nice if implemented into library.

I used the code below for feather MO as interim

void updateRTC() {  //Sets Internal  M0 RTC from Structured ASCII msg from GSM Module
  char buffer[23];
  fona.getTime(buffer, 23);
  rtc.setYear( ((buffer[1] - 48) * 10) + (buffer[2] - 48) );
  rtc.setMonth( ((buffer[4] - 48) * 10) + (buffer[5] - 48) );
  rtc.setDay( ((buffer[7] - 48) * 10) + (buffer[8] - 48) );
  rtc.setHours( ((buffer[10] - 48) * 10) + (buffer[11] - 48) );
  rtc.setMinutes( ((buffer[13] - 48) * 10) + (buffer[14] - 48) );
  rtc.setSeconds( ((buffer[16] - 48) * 10) + (buffer[17] - 48) );
}

jwallis, 2020-02-04:
Check https://github.com/botletics/SIM7000-LTE-Shield it might have better RTC support.

OscarVanL, 2020-04-04:
This PR from 2016 implements it and works perfectly for me on the FONA 3G. All it needs is to be merged in.