adafruit/Adafruit_FONA #114: Enable Network Time Sync is not implemented for FONA 3G

OscarVanL, 2020-04-03:
Hi,

The Adafruit FONA library has a function enableNetworkTimeSync(), however, this does not work for the FONA 3G as the AT+CLTS command this uses is not used by FONA 3G.

This has been a source of frustration for me over the past days, as the [module's documented GPS week rollover issue](https://cdn-learn.adafruit.com/assets/assets/000/084/848/original/GPS-week-rollover_Simcom.pdf?1574619386) and inability to enable network time sync has left me without a usable date/time.

Fortunately, I've worked out what AT commands are required to enable this for FONA 3G. I've made a thread on the Adafruit Forums about this [here](https://forums.adafruit.com/viewtopic.php?f=54&t=164135).

I hope to make a pull request in the near future that adds support for this feature on the FONA 3G to Adafruit FONA. If I don't get around to it, feel free to do so for me :)

OscarVanL, 2020-04-03:
I've made the PR to implement Network Time Sync on the FONA 3G.

Nathan-ma, 2020-04-21:
Great Oscar! can I ask you how can I use this?

What I'm expecting to receive is a NTP Number from the board, I've noticed your implementation enables the NTPSync, How can I receive the output from it? and how can I set the server?

Should I use it like in this example? 

[image]

Thank you and
Sorry in advance for the bad english.