adafruit/Adafruit_FONA #116: TCPconnected Does not work on FONA 3G

OscarVanL, 2020-04-07:
The TCPconnected function sends `AT+CIPSTATUS`, however this is not a valid command for the FONA 3G's SIM5320. As a result, this fails every time.

In some other libraries like Adafruit_MQTT_FONA it checks the TCP connection to the MQTT server is open with TCPconnected(), this fails every time for the above issue.

I have not yet found which AT command is a substitute for AT+CIPSTATUS for the FONA 3G.

kukukk, 2020-04-14:
Hi,

If you are using only 1 active connection at a time, you should try something like this:
```
boolean Adafruit_FONA_3G::TCPconnected(void) {
  float index;
  return sendParseReply(F("AT+CIPOPEN?"), F("+CIPOPEN: "), &index, ',', 4);
}
```

Btw, are the TCP functions working for you with FONA 3G?

OscarVanL, 2020-04-14:
That looks like it should work :)

The TCP functions in this fork do not work on the FONA 3G without modification, as it uses different AT conmands. I adapted them and have got it to work. I can open a TCP socket successfully and send TCP packets. I also got the MQTT Connect and Publish functionality to work too.

I've committed some of the changes to my fork of this repo, if you're interested you can have a look.

kukukk, 2020-04-14:
They say that the TCP functions are working with FONA 3G: https://forums.adafruit.com/viewtopic.php?f=54&t=162392

But I had to implement the following TCP functions to be able to work with FONA 3G:
```
bool Adafruit_FONA_3G::TCPconnect(char *server, uint16_t port)
{
  flushInput();

  // Close old connection
  getReply(F("AT+CIPCLOSE=0"));

  // Get data manually
  if (! sendCheckReply(F("AT+CIPRXGET=1"), ok_reply) ) return false;

  char sendcmd[50];
  sprintf(sendcmd, "AT+CIPOPEN=0,\"TCP\",\"%s\",%u", server, port);

  return sendCheckReply(sendcmd, F("Connect ok"), 5000);
}
```
```
bool Adafruit_FONA_3G::TCPclose(void)
{
  return sendCheckReply(F("AT+CIPCLOSE=0"), ok_reply);
}
```
```
boolean Adafruit_FONA_3G::TCPconnected(void) {
  float index;
  return sendParseReply(F("AT+CIPOPEN?"), F("+CIPOPEN: "), &index, ',', 4);
}
```
```
boolean Adafruit_FONA_3G::TCPsend(char *packet, uint8_t len) {
  flushInput();

  char sendcmd[20];
  sprintf(sendcmd, "AT+CIPSEND=0,%u", len);
  if (! sendCheckReply(sendcmd, F(">"))) return false;

  return sendCheckReply(packet, ok_reply);
}
```
```
uint16_t Adafruit_FONA_3G::TCPavailable(void) {
  float avail;

  if (! sendParseReply(F("AT+CIPRXGET=4,0"), F("+CIPRXGET: 4,0,"), &avail, ',', 0) ) return 0;

  DEBUG_PRINT (avail); DEBUG_PRINTLN(F(" bytes available"));

  return (uint16_t)avail;
}
```
```
uint16_t Adafruit_FONA_3G::TCPread(uint8_t *buff, uint8_t len) {
  float avail;

  mySerial->print(F("AT+CIPRXGET=2,0,"));
  mySerial->println(len);
  readline();

  if (! parseReply(F("+CIPRXGET: 2,0,"), &avail, ',', 0)) return false;

  readRaw((uint16_t)avail);

#ifdef ADAFRUIT_FONA_DEBUG
  DEBUG_PRINT (avail); DEBUG_PRINTLN(F(" bytes read"));
  for (uint8_t i=0;i<avail;i++) {
  DEBUG_PRINT(F(" 0x")); DEBUG_PRINT(replybuffer[i], HEX);
  }
  DEBUG_PRINTLN();
#endif

  memcpy(buff, replybuffer, (uint16_t)avail);

  return (uint16_t)avail;
}
```

OscarVanL, 2020-04-14:
Looks similar to (but better/more complete) than the stuff I had to do. Wish I'd had those functions 1 week ago today, would have saved me a lot of time.

You should make a PR to add this support for the FONA 3G :)

TheRobertTalley, 2021-08-06:
This is exactly what im looking for! though implementing it is a bit beond me at the moment. i am using Tim woo's botletics library trying to connect to AIO.

what am i doing wrong?

`
#include <Adafruit_FONA.h> // https://github.com/botletics/SIM7000-LTE-Shield/tree/master/Code
#include <Adafruit_MQTT.h>
#include <Adafruit_MQTT_Client.h>
#include <Adafruit_MQTT_FONA.h>
#include <WiFiClientSecure.h>

#define SIMCOM_7000            //pinset for a lilygo tcallsim7000 and external gps
#define FONA_PWRKEY 5
#define FONA_RST 4
#define FONA_TX 26 // For GSM
#define FONA_RX 27 
#define TXPin 32 // For GPS module 
#define RXPin 33 
#define SD_MISO     2
#define SD_MOSI     15
#define SD_SCLK     14
#define SD_CS       13
#define LED_PIN     12
#define BAT_ADC     35
#define SOLAR_ADC   36
#define GPSBaud   9600

uint32_t x=0; //formqqttoconnect
/*****************************************************MODEM***********************************/
#include <HardwareSerial.h>
HardwareSerial fonaSS(1);
Adafruit_FONA_LTE fona = Adafruit_FONA_LTE();//woo modem marker
uint8_t readline(char *buff, uint8_t maxbuff, uint16_t timeout = 0);
uint8_t type;
char replybuffer[255]; // this is a large buffer for replies
char imei[16] = {0}; // MUST use a 16 character buffer for IMEI!
/************************* WiFi Access Point *********************************

#define WLAN_SSID "ATTn"       //Guest Wifi 
#define WLAN_PASS "faec" 

/************************* Adafruit.io Setup *********************************/

#define AIO_SERVER      "io.adafruit.com"

// Using port 8883 for MQTTS
#define AIO_SERVERPORT  1883  //1883

// Adafruit IO Account Configuration
// (to obtain these values, visit https://io.adafruit.com and click on Active Key)
#define AIO_USERNAME  "namehere"
#define AIO_KEY       "keyhere"

/************ Global State (you don't need to change this!) ******************/
// WiFiFlientSecure for SSL/TLS support
WiFiClientSecure client;

// io.adafruit.com root CA
const char* adafruitio_root_ca = \
    "-----BEGIN CERTIFICATE-----\n" \
    "MIIDrzCCApegAwIBAgIQCDvgVpBCRrGhdWrJWZHHSjANBgkqhkiG9w0BAQUFADBh\n" \
    "MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3\n" \
    "d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBD\n" \
    "QTAeFw0wNjExMTAwMDAwMDBaFw0zMTExMTAwMDAwMDBaMGExCzAJBgNVBAYTAlVT\n" \
    "MRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxGTAXBgNVBAsTEHd3dy5kaWdpY2VydC5j\n" \
    "b20xIDAeBgNVBAMTF0RpZ2lDZXJ0IEdsb2JhbCBSb290IENBMIIBIjANBgkqhkiG\n" \
    "9w0BAQEFAAOCAQ8AMIIBCgKCAQEA4jvhEXLeqKTTo1eqUKKPC3eQyaKl7hLOllsB\n" \
    "CSDMAZOnTjC3U/dDxGkAV53ijSLdhwZAAIEJzs4bg7/fzTtxRuLWZscFs3YnFo97\n" \
    "nh6Vfe63SKMI2tavegw5BmV/Sl0fvBf4q77uKNd0f3p4mVmFaG5cIzJLv07A6Fpt\n" \
    "43C/dxC//AH2hdmoRBBYMql1GNXRor5H4idq9Joz+EkIYIvUX7Q6hL+hqkpMfT7P\n" \
[... 211 lines cut ...]

void GPRSon(){
  // turn GPRS on
  if (!fona.enableGPRS(true)){
   Serial.println(F("Failed to turn on"));
   delay(2000);
   GPRSon();
  }
}
void RootCA() {
  fona.addRootCA(adafruitio_root_ca);
  Serial.println("sert is on");
}
void TCPc() {
  char sendcmd[50];
  sprintf(sendcmd, "AT+CIPOPEN=0,\"TCP\",\"%s\",%u", AIO_SERVER, AIO_SERVERPORT);
  fonaSS.print(sendcmd);
  Serial.println("TCP?");
delay(5000);
}`

i have posted this here as well:
https://www.reddit.com/r/esp32/comments/oykl6r/adafruit_io_esp32sim7000_lilygo_sim7000x_aio_over/

any help or examples would get you loved and appreciated.

TheRobertTalley, 2021-08-06:
This is what im looking for! im using tim woo's botletics library, esp32 and a sim7000 trying to connect to AIO.
any ideas what is wrong with this?

https://github.com/TheRobertTalley/lilygo-sim7000-ESP32/commit/cde2727706193dbcb42b19255d14c1e45d8da83f

I dont know how to use what kukukk posted directly. An example would be well appreciated, deffinatley a free beer in order. Im not sure how to call functions from the library. I have posted here also if the nice folks here would take a look:

https://www.reddit.com/r/esp32/comments/oykl6r/adafruit_io_esp32sim7000_lilygo_sim7000x_aio_over/