adafruit/Adafruit_FONA #131: SIM7070 fails when adding body to HTTP POST

KnightOfNih, 2021-04-13:
- Arduino board:  [Waveshare SIM7070G Cat-M/NB-IoT/GPRS HAT](SIM7070G Cat-M/NB-IoT/GPRS HAT) 

- Platform.io 
- 
Version: 1.55.0
Commit: c185983a683d14c396952dd432459097bc7f757f
Date: 2021-03-30T16:01:05.981Z (1 wk ago)
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Darwin x64 18.7.0

- Steps to Repro.  Sketch attached.  Uses the Adafruit_FONA library and based on the examples provided for Http POST.

Hello Gents,

First off, I'd like to give a very greatful thank you to the library writers for doing this.  There's no way I would have gotten as far as I would have without this very helpful repo.  Thank you.

Using the example sketches provided with the library to do an HTTP POST.  A POST operation *works* (needed to add some delay commands for SIM7070) but when I add a body to the POST, the POST fails.

In particular, here are the commands to add the body post.

`void addBodyToPost()
{
  // Sending plan text and wait one second, b/c I've seen that in other tutorials
  fona.HTTP_addHeader("Content-Type", "text/plain", 10);
  delay(1000);

  fona.HTTP_addHeader("Host", "b225f0b114f9.ngrok.io", 21);
  delay(1000);

  fona.HTTP_addHeader("User-Agent", "IOE Client", 10);
  delay(1000);

  // Add the SHBOD.  It will return a ">" when it is ready for data
  char cmdBuff[150];
  char body[] = {"foo"};
  int bodySize = sizeof(body) / sizeof(body[0]);

  // Format your command and wait up to 2 seconds for SIM7070 to respond
  sprintf(cmdBuff, "AT+SHBOD=%i,10000", bodySize);
  fona.getReply(cmdBuff, 2000);

  // If you get a reply, hope it's a prompt
  if (strstr(fona.replybuffer, ">") == NULL)
    Serial.println("No > in reply buffer");

  // Send the data to the module and wait for OK
  if (!fona.sendCheckReply(body, "OK", 2000))
    Serial.println("No OK in reply buffer.");

  // Data added!  Ha.  That's what they WANT you to think.
}`

Here is the entire sketch below.  This was modified slightly from the ESP32 example.  Again, everything works perfectly when I comment out the "addBodyToPost" function.

`#include <Arduino.h>
#include "Adafruit_FONA.h" // https://github.com/botletics/SIM7000-LTE-Shield/tree/master/Code

#define SIMCOM_7070 // I'm using one of these

#define FONA_PWRKEY 18
#define FONA_RST 5
#define FONA_TX 16 // ESP32 hardware serial RX2 (GPIO16)
#define FONA_RX 17 // ESP32 hardware serial TX2 (GPIO17)

// For ESP32 hardware serial
#include <HardwareSerial.h>
HardwareSerial fonaSS(1);
[... 228 lines cut ...]
        ---> AT+SHCONN
        <--- OK
        ---> AT+SHSTATE?
        <--- +SHSTATE: 1
        ---> AT+SHCHEAD
        <--- OK
        ---> AT+SHAHEAD="Content-Type","text/plain"
        <--- OK
        ---> AT+SHAHEAD="Host","b225f0b114f9.ngrok.io"
        <--- OK
        ---> AT+SHAHEAD="User-Agent","IOE Client"
        <--- OK
        ---> AT+SHBOD=4,10000
        <--- > 
        ---> foo
        <--- OK
        ---> AT+SHREQ="http://b225f0b114f9.ngrok.io",3
        <--- OK
Unable to get status
        ---> AT+SHDISC
        <--- AT+SHDISC_

Any ideas on what's going wrong?

I've tried with and without headers.  I've tried with varying SHBOD lengths (i.e. AT+SHBOD=3,... instead of AT_SHBOD=4).  No luck.

ladyada, 2021-04-13:
hi please ask waveshare for help & support - we dont own that module! :)

closed 2021-04-13 by ladyada

KnightOfNih, 2021-04-13:
Well shucks.  Does your team have an example of using HTTP POST with a body attached?  In theory, it should all work the same since you both use a SIM7070 (I would assume).

ladyada, 2021-04-13:
nope, we have never used a SIM7070, don't even own one!

KnightOfNih, 2021-04-13:
Ok, thank you for the prompt responses.  Sorry, was using a library built on top of yours!  https://github.com/botletics/SIM7000-LTE-Shield